jonasjuffinger/TimeLapse #2: Not working in RX100M3

softicewong, 2017-10-31:
It is not working neither for firmware V1.2 nor V2.0.

It just turn off camera and turn it on again.

jonasjuffinger, 2017-10-31:
It should create a folder named TIMELPS on the sdcard, with a file LOG.TXT in it.
Could you please paste the contents of this file.

softicewong, 2017-10-31:
[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.setSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
	at android.app.Activity.performResume(Activity.java:3832)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.setSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
	at android.app.Activity.performResume(Activity.java:3832)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.setSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
	at android.app.Activity.performResume(Activity.java:3832)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.setSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
[... 136 lines cut ...]
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.setSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
	at android.app.Activity.performResume(Activity.java:3832)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

jonasjuffinger, 2017-10-31:
Ok I see the problem, if I find time tomorrow I will fix it and upload a new version

jonasjuffinger, 2017-11-01:
Please try this https://github.com/jonasjuffinger/TimeLapse/releases/download/v0.3.1/TimeLapse-release-0.3.1.apk

softicewong, 2017-11-02:
After the first installation, it shows "Recovering Data, Please wait...", then turn the camera off. For the 2nd run and after, it just turn off and turn on again. Below is the LOG.txt

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.isSupportedSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
	at android.app.Activity.performResume(Activity.java:3832)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.isSupportedSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
	at android.app.Activity.performResume(Activity.java:3832)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.isSupportedSilentShutterMode
	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
	at android.app.Activity.performResume(Activity.java:3832)
	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:130)
	at android.app.ActivityThread.main(ActivityThread.java:3683)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:507)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
	at dalvik.system.NativeStart.main(Native Method)

jonasjuffinger, 2017-11-02:
Ok it's still a very similar problem, I hoped the camera would not be so dump. I will fix it.

jonasjuffinger, 2017-11-02:
Please try version 0.3.2

softicewong, 2017-11-02:
0.3.2 is working fine for my camera RX10M3 (firmware version 2.0).

Thank you very much.

jonasjuffinger, 2017-11-03:
That's good to hear!

closed 2017-11-03 by jonasjuffinger

jonasjuffinger, 2018-06-03:
Hi @softicewong I hope to reach you like this. You are also using this app with the RX100M3 do you also have this problem?
https://github.com/jonasjuffinger/TimeLapse/issues/15
Thanks for your anwser!

terragady, 2018-06-03:
I am using it with RX100M3  without problems. Not much settings but it works like it should