jonasjuffinger/TimeLapse #4: Not working in a6000

wlodik, 2017-11-02:
Hello.
It doesn't work on my a6000 firmware 3.20 .
I can install and open version 0.3, but after I set up all things and push start button it looks like it's turning off and on, and that's all.
Here is log file:

> [INFO] Hello World
> [INFO] onCreate
> [INFO] onResume
> [ERROR] java.lang.NoSuchMethodError: com.sony.scalar.hardware.CameraEx$ParametersModifier.isSupportedSilentShutterMode
> 	at com.jonasjuffinger.timelapse.ShootActivity.onResume(SourceFile:119)
> 	at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1150)
> 	at android.app.Activity.performResume(Activity.java:3832)
> 	at android.app.ActivityThread.performResumeActivity(ActivityThread.java:2110)
> 	at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:2135)
> 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:1668)
> 	at android.app.ActivityThread.access$1500(ActivityThread.java:117)
> 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:931)
> 	at android.os.Handler.dispatchMessage(Handler.java:99)
> 	at android.os.Looper.loop(Looper.java:130)
> 	at android.app.ActivityThread.main(ActivityThread.java:3683)
> 	at java.lang.reflect.Method.invokeNative(Native Method)
> 	at java.lang.reflect.Method.invoke(Method.java:507)
> 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:839)
> 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:597)
> 	at dalvik.system.NativeStart.main(Native Method)
> 

I tried to install version 0.3.1 but in pcma-gui v.015 i see this:

> Loading app list
> Found 8 apps
> 
> Using drivers Windows-MSC, Windows-MTP
> Looking for Sony devices
> 
> Querying MTP device
> Sony Corporation ILCE-6000 is a camera in MTP mode
> 
> Model:              ILCE-6000
> Product code:       0032826830
> Serial number:      04150474
> Firmware version:   3.20
> Lens:               Model 0x12a08019 (Firmware 1.09)
> 
> Using drivers Windows-MSC, Windows-MTP
> Looking for Sony devices
> 
> Querying MTP device
> Sony Corporation ILCE-6000 is a camera in MTP mode
> 
> Switching to app install mode
> Waiting for camera to switch...
> Starting task
> Starting communication
> Uploading 0%
> Uploading 100%
> Downloading 100%
> Installing 0%
> Traceback (most recent call last):
>   File "C:\projects\sony-pmca-re\pmca-gui.py", line 77, in do
>   File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 255, in installCommand
>   File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 75, in installApp
> Exception: Communication error 100: Error completed
>

jonasjuffinger, 2017-11-02:
Thanks for reporting. I'm already working on a fix for this problem.
I added the issue to the readme file.

jonasjuffinger, 2017-11-02:
Please try version 0.3.2

wlodik, 2017-11-02:
It works now. Thank you. I'll test all options tomorrow.

jonasjuffinger, 2017-11-02:
Perfect!
The silent shutter option in the second tab does obviously not work. I will hide this option in the next version.
More options do not exist, but I'm open for suggestions.

closed 2017-11-02 by jonasjuffinger