jonasjuffinger/TimeLapse #5: Nex 5R Issues

quikee, 2017-11-14:
I tried it with my Nex 5R - the timelapse works but there are issues: 
- Silent shutter doesn't work, which is understandable - maybe hide the option if it is known to not work on a specific model.
- Setting the time interval is 1, 5, 9s - I guess only on this camera, maybe a more limited set would be better (maybe 1,2,3,4,5,10,15,20,30,60,...) Probably 1,5,9 sequence has to do something with the UI element
- Would be nice if time and number of frames selection could be done with the scroll wheel in addition to left/right button (it's much faster)
- Even when setting it to 1s, the actual time is more like 2s according to metadata (shutter speed was 1/80 so that shouldn't be a problem)
- After a while it starts shooting more pics per second (up to 6 pics / second), it is like it gets out of sync -> this is the most annoying issue I found

jonasjuffinger, 2017-11-14:
Thank you for this report.

Hiding the silent shutter option was planned for the next release

You have these large steps because the scroll wheel doesn't work. On the a6300 the scroll wheel works and the steps are like you proposed. 

Do you have picture preview enabled? Is the timing better if it is disabled?

This weeks I won't find time, but next week I will build you a version that gathers additional information about the hardware to include the scroll wheel. And i will look into this timing issue.

jonasjuffinger, 2017-12-21:
I'm sorry I couldn't find time to look into this with more detail. Could you please try with the attached apk if settings the time interval works with any of the dials.

[TimeLapse-release-0.3.3.apk.zip](https://github.com/jonasjuffinger/TimeLapse/files/1578120/TimeLapse-release-0.3.3.apk.zip)

quikee, 2017-12-24:
No problem, there is no immediate rush. I was also very busy so I didn't play with this much. 

I tried once again just after my initial comment and with disabling preview and more setting more things to manual, the interval between images was more precise, but still not very exact, after a while it still started making multiple pictures in a row. Next time I tried to reproduce this, but it again made previews even when I disabled them in settings and the interval was way worse - it sometimes buffered up and started making multiple pictures in a row (like up to 6 in less than a sec.)

I have some time in the following week so I'll try the APK and I'll report back how it works.

Thanks.

n0credit, 2017-12-27:
I tried the original version in a a6000 and it worked fine. Interval was spot on. Will do a more extensive test in the coming days but I guess the ax000 have a lot in common so it works pretty much the same, and thanks for this!

jonasjuffinger, 2017-12-27:
@quikee The App creates the TIMELPS/LOG.TXT. Could you please post it, maybe it contains some useful information. As @n0credit wrote there is no problem on the a6000 or my a6300 so I can not reproduce this problem.
Thanks

jonasjuffinger, 2018-03-16:
Closed due to inactivity

closed 2018-03-16 by jonasjuffinger