jonasjuffinger/TimeLapse #6: 1-5-9 Intervals on A7Rii

drc13a, 2017-12-20:
Hi,

Just installed your app and am excited to use it while overseas next week. Unfortunately the interval steps are quite larger (1-5-9 etc) and having read the other issue it appears to be wheel related (it doesn't do anything when I spin the rear wheel) I control the settings by pressing right or left.

Is it possible to activate the wheel for the A7Rii or as a stop gap measure make the left/right presses include all the increments rather than the bigger jumps?

Thank in advance!

jonasjuffinger, 2017-12-21:
Hi, could you please try with the attached apk if any of the dials work.

[TimeLapse-release-0.3.3.apk.zip](https://github.com/jonasjuffinger/TimeLapse/files/1578122/TimeLapse-release-0.3.3.apk.zip)

drc13a, 2017-12-21:
I can happily report that the attached apk let's me fully control both of the sliders using the dial meaning I get all the small increments. Thank you very much! :)

jonasjuffinger, 2017-12-21:
That's good to hear!
That's still strange, but if the main wheel works it's ok for now.

closed 2017-12-21 by jonasjuffinger