jonasjuffinger/TimeLapse #8: timesettings:  several steps between 1s and 5s

hubertmeier, 2017-12-30:
This is a very usefull app, would it be possible to have some steps between 1s and 5s (1.5s, 2s, 3s, 5s). When writing RAW, 1s is too short ...

Thank you for your work

jonasjuffinger, 2018-01-01:
These steps are selectable when you use one of the dials. What camera do you have?

hubertmeier, 2018-01-02:
Oh, i didn't know that, i tested only the left-right buttons ... i will try it

I tested width NEX-6, A5100, and will test it with A7 series, RX1

Would it be possible to show something on the display between the frame capture like the number and / or the time of the remaining frames ?

hubertmeier, 2018-01-03:
OK, i tested it

NEX-6: the slider works (1s, 5s ...), the wheels both do not work
A5100: the slider and the rotating keypad on the back work

hubertmeier, 2018-01-03:
Oh, i did not want to close this issue ... sorry

jonasjuffinger, 2018-06-03:
I just released a new version of the app, could you try it again?

closed 2020-01-03 by jonasjuffinger