jonasjuffinger/TimeLapse #9: Doesn't work with a6500 (firmware 1.04)

wojciechpolak, 2018-01-03:
TIMELPS/LOG.TXT:

```
[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[INFO] 2
[INFO]   Remaining Time: -465998
[ERROR] java.lang.RuntimeException: burstableTakePicture() failed
        at com.sony.scalar.hardware.CameraEx.native_burstableTakePicture(Native Method)
        at com.sony.scalar.hardware.CameraEx.burstableTakePicture(CameraEx.java:1042)
        at com.jonasjuffinger.timelapse.ShootActivity.J(SourceFile:201)
        at com.jonasjuffinger.timelapse.ShootActivity.g(SourceFile:20)
        at com.jonasjuffinger.timelapse.ShootActivity$1.run(SourceFile:61)
        at android.os.Handler.handleCallback(Handler.java:672)
        at android.os.Handler.dispatchMessage(Handler.java:93)
        at android.os.Looper.loop(Looper.java:137)
        at android.app.ActivityThread.main(ActivityThread.java:4745)
        at java.lang.reflect.Method.invokeNative(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:511)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
        at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[INFO] 2
[INFO]   Remaining Time: -110054
[ERROR] java.lang.RuntimeException: burstableTakePicture() failed
        at com.sony.scalar.hardware.CameraEx.native_burstableTakePicture(Native Method)
        at com.sony.scalar.hardware.CameraEx.burstableTakePicture(CameraEx.java:1042)
        at com.jonasjuffinger.timelapse.ShootActivity.J(SourceFile:201)
        at com.jonasjuffinger.timelapse.ShootActivity.g(SourceFile:20)
        at com.jonasjuffinger.timelapse.ShootActivity$1.run(SourceFile:61)
        at android.os.Handler.handleCallback(Handler.java:672)
        at android.os.Handler.dispatchMessage(Handler.java:93)
        at android.os.Looper.loop(Looper.java:137)
        at android.app.ActivityThread.main(ActivityThread.java:4745)
        at java.lang.reflect.Method.invokeNative(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:511)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
        at dalvik.system.NativeStart.main(Native Method)
```

jonasjuffinger, 2018-06-03:
Sorry I totally overlooked this report. Did you manage to get it working? I just released a new version of the app, it could solve the problem.

wojciechpolak, 2018-06-09:
Yes, it started working with v1.0.0! Thank you.

But unfortunately it doesn't work with Continuous Bracket mode (in Manual mode).
It just ignores the Cont. Bracket mode and shoots single photos.
Is it this app limitation or the camera?

jonasjuffinger, 2018-06-10:
Happy to hear that!

It's true that it doesn't work with Cont. Bracket mode. I will look into that for the next version.

jonasjuffinger, 2019-07-18:
Should be fixed with the newest version

closed 2019-07-18 by jonasjuffinger