jonasjuffinger/TimeLapse #10: RX100 M4 - On/OFF Issue

shinysidez, 2018-02-01:
When you hit Start - to start the time lapse - The camera switches OFF and ON and no pictures are taken. I believe this issue was reported in the past by someone else - It is doing it again on the latest release. Here's the contents of the log file

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[INFO] 0
[INFO]   Remaining Time: -177197
[ERROR] java.lang.RuntimeException: burstableTakePicture() failed
	at com.sony.scalar.hardware.CameraEx.native_burstableTakePicture(Native Method)
	at com.sony.scalar.hardware.CameraEx.burstableTakePicture(CameraEx.java:1033)
	at com.jonasjuffinger.timelapse.ShootActivity.J(SourceFile:201)
	at com.jonasjuffinger.timelapse.ShootActivity.g(SourceFile:20)
	at com.jonasjuffinger.timelapse.ShootActivity$1.run(SourceFile:61)
	at android.os.Handler.handleCallback(Handler.java:672)
	at android.os.Handler.dispatchMessage(Handler.java:93)
	at android.os.Looper.loop(Looper.java:137)
	at android.app.ActivityThread.main(ActivityThread.java:4745)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:511)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[INFO] 0
[INFO]   Remaining Time: -85407
[ERROR] java.lang.RuntimeException: burstableTakePicture() failed
	at com.sony.scalar.hardware.CameraEx.native_burstableTakePicture(Native Method)
	at com.sony.scalar.hardware.CameraEx.burstableTakePicture(CameraEx.java:1033)
	at com.jonasjuffinger.timelapse.ShootActivity.J(SourceFile:201)
	at com.jonasjuffinger.timelapse.ShootActivity.g(SourceFile:20)
	at com.jonasjuffinger.timelapse.ShootActivity$1.run(SourceFile:61)
	at android.os.Handler.handleCallback(Handler.java:672)
	at android.os.Handler.dispatchMessage(Handler.java:93)
	at android.os.Looper.loop(Looper.java:137)
	at android.app.ActivityThread.main(ActivityThread.java:4745)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:511)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[INFO] 0
[INFO]   Remaining Time: -93275
[ERROR] java.lang.RuntimeException: burstableTakePicture() failed
	at com.sony.scalar.hardware.CameraEx.native_burstableTakePicture(Native Method)
	at com.sony.scalar.hardware.CameraEx.burstableTakePicture(CameraEx.java:1033)
	at com.jonasjuffinger.timelapse.ShootActivity.J(SourceFile:201)
	at com.jonasjuffinger.timelapse.ShootActivity.g(SourceFile:20)
	at com.jonasjuffinger.timelapse.ShootActivity$1.run(SourceFile:61)
	at android.os.Handler.handleCallback(Handler.java:672)
	at android.os.Handler.dispatchMessage(Handler.java:93)
	at android.os.Looper.loop(Looper.java:137)
	at android.app.ActivityThread.main(ActivityThread.java:4745)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:511)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
	at dalvik.system.NativeStart.main(Native Method)

[INFO] Hello World
[INFO] onCreate
[INFO] onResume
[INFO] 0
[INFO]   Remaining Time: -107031
[ERROR] java.lang.RuntimeException: burstableTakePicture() failed
	at com.sony.scalar.hardware.CameraEx.native_burstableTakePicture(Native Method)
	at com.sony.scalar.hardware.CameraEx.burstableTakePicture(CameraEx.java:1033)
	at com.jonasjuffinger.timelapse.ShootActivity.J(SourceFile:201)
	at com.jonasjuffinger.timelapse.ShootActivity.g(SourceFile:20)
	at com.jonasjuffinger.timelapse.ShootActivity$1.run(SourceFile:61)
	at android.os.Handler.handleCallback(Handler.java:672)
	at android.os.Handler.dispatchMessage(Handler.java:93)
	at android.os.Looper.loop(Looper.java:137)
	at android.app.ActivityThread.main(ActivityThread.java:4745)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:511)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
	at dalvik.system.NativeStart.main(Native Method)

jonasjuffinger, 2018-02-01:
Thank you for reporting. I already know of this issue, I was very busy the last month but I hope to find time in the near future to look into it.

shinysidez, 2018-02-02:
What is weird, is that it was working and then I updated to the new v.3.3 and this camera On/Off error started happening. So I went back to the v3.2 and no luck at all. Then I tried basically all versions 3.1, 3.2 and 3.3 and they do the ON/OFF. I really liked your app for it's simplicity and would rather pay you for this app than Sony. I hope you can resolve this.

jonasjuffinger, 2018-02-03:
Ok that is really weird

shinysidez, 2018-02-05:
Here's what I found and kind of resolved the issue
For RX100 M4

1) The silent shutter needs to be disabled or unchecked in the app - It would be nice if this was on the same tab/page as the rest of the settings
2) The app will not run if you are set or self time or continous shooting - Only Single shoot will work.

So far it's running ok

jonasjuffinger, 2018-02-05:
Thank you for that finding!

I will try to change this settings directly in the app, so that the user doesn't have to do it.

shinysidez, 2018-02-06:
Jonas,
Thank you for making the app. Here's some recommendations
1) Update the Readme with this issue - if you are unable to update the app currently
2) In the future versions
a) Make the silent shutter option on the same page as the time lapse settings - One tab for everything
b) Have it unchecked by default
c) I don't know how to resolve the "If self timer/continous setting is enabled, then deactivate it for TimeLapse app or how to show an error if it is active"

Other than that - Your app is much better than the Sony TimeLapse app for sure - simple and gets the job done without the unnecessary flashy shit

closed 2018-02-06 by shinysidez