jonasjuffinger/TimeLapse #11: Instalation impossible on Sony A7ii

French-People, 2018-02-14:
Hello
I try many solutions to Instaler "TimeLapse by Jonasjuffinger " but nothing works for me. Otherwise OpenMemories: Tweak works correctly with installation of PMCA-GUI-v 0.16-win. EXE under win 10.
Here's the news:

Using Windows Drivers-MSC, Windows-MTP
Looking for Sony devices
Querying MTP Device
Sony Corporation ILCE-7M2 is a camera in MTP mode
Model: ILCE-7M2
Product Code: XXXXXXXXXX
Serial Number: xxxxxxxx
Firmware Version: 4.00

Thank you for your incredible work and the time you offer to the whole world:) The French tell you a big thank you!

French-People, 2018-02-14:
Downloading apk
Starting task
Starting communication
Uploading 0%
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 74, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 256, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 74, in installApp
  File "C:\projects\sony-pmca-re\pmca\installer\__init__.py", line 84, in install
ConnectionRefusedError: [WinError 10061] Aucune connexion n’a pu être établie car l’ordinateur cible l’a expressément refusée

jonasjuffinger, 2018-02-21:
Thats very strange, do other Apps like FocusBracket also work?

French-People, 2018-03-12:
I find the problem, it's with my configuration of my Proxy ! Now it's ok and your work is perfect ! Thank you very much, you are the best ! I share your work on facebook if you approuve !

jonasjuffinger, 2018-03-12:
I'm happy you found the problem. Thank you very much, it would be really great if you share it!

closed 2018-03-12 by jonasjuffinger

pvmodin, 2018-08-06:
I'm unable to install V1.0.0 on RX100V
Tried it from pmca-gui 0.17 app list and manual install from github apk
v0.3.3 worked from github apk
Intalled FocusBracket and pmcaDemo just fine from app list.

Package: com.jonasjuffinger.timelapse
Version: 1.0.0
Starting task
Starting communication
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 79, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 291, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 84, in installApp
  File "C:\projects\sony-pmca-re\pmca\installer\__init__.py", line 48, in install
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 735, in emptyBuffer
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 702, in receive
Exception: Unknown rest message type: 0x504f

Any help getting v1.0.0 to install? I have no proxies to disable like previously mentioned.