jonasjuffinger/TimeLapse #13: [REQUEST] Add Exposure Time

bcdub, 2018-03-11:
Okay hear me out. This has amazing potential for milky way photography. The A7/R/S and the A6000/A5100 all suffer from the star eater issue in bulb mode. If you could add the option to set exposure time (through T mode, not bulb [B] mode) and added options for 1,2,4,8,15 minutes, then these cameras would once again be relevant for astrophotography. Please consider it and thank you for your time.

jonasjuffinger, 2018-03-11:
That would be really great, but Sony does this filtering directly in the firmware so there is no way for an app to disable or evade it. The only thing we can do is increase the pressure on Sony to add a menu option to disable it.
https://www.change.org/p/sony-remove-star-eater-en

closed 2018-03-11 by jonasjuffinger

bcdub, 2018-03-12:
You are correct, but the A7/S/R only does this in bulb mode, so my hypothesis was that if an app could initiate a time through T mode that it would circumvent the filtering. I'm pretty sure it would work. If you are of the opinion that you are unable to initiate a different exposure time than the ones preset in T mode then I agree that an app has no chance. 

Thank you for your response.