jonasjuffinger/TimeLapse #17: Wrong interval on A6300, 5s shortest?

rlesac, 2018-05-23:
I could not get shorter interval then 5 sec for RAW, even though the app is set to 1s.

Settings: manual focus, aperture and shutter, silent shooting
SD card: Samsung Pro 80MB write speed u3

**Timelapse app 1s interval:**
RAW:
single shoot:  14 photos per minute ~ 4.3s interval
3 bracket: 42 photos, 14 brackets per minute ~ 4.3s interval

JPG:
single shoot: 15 photos per minute ~ 4s interval

**Timelapse app 2s interval:**
RAW:
single shoot: 10 photos per minute ~ 6s interval

-----------------
**Wired remote (Continuous low):**

RAW: 78 photos per minute  ~ 0.76s interval (this also seems to be the limit for the camera SD interface as it's close to 30MB/s writes)

JPG: 97 photos per minute ~ 0.6s interval (this would seem to be the processing limit, as it's only 19MB/s of writes)
-----------------------

Those are sustained numbers for the wired remote, measured minutes into the timelapse.

It would seem that there is something fishy going on with the interval and or sleep?

jonasjuffinger, 2018-06-03:
That's very strange. I also have a a6300 at it always worked for me.
Could you please try the newest version i released today and if the problem persist upload the file TIMELPS/LOG.TXT

rlesac, 2018-07-31:
Just tested out the new version and heres the results for me:
RAW: 54 ppm
JPG: 59 ppm

It's much better.

I'm guessing you're waiting for the photo to be saved, before starting the counter for the new one.

Bracketing stopped working in the new version.

rlesac, 2018-09-06:
Would you please consider adding another setting, fastest, so the camera will just shoot as fast as possible?
Thanks.

jonasjuffinger, 2020-01-03:
Done, see burst mode

closed 2020-01-03 by jonasjuffinger