jonasjuffinger/TimeLapse #18: not keeping camera settings after 1st shot

Thaone73, 2018-07-10:
Hi, I am using sony a6500. I have registered my timelapse settings (WB, ISO, etc,) to one of the 2 customized settings on the main dial. Before doing the timelapse, I select these preset camera settings, then I check the settings are set properly by taking pictures. Then I launch the timelapse. The 1st pic is taken using the chosen camera settings, but all subsequent pics are taken using some kind of Auto settings (AWB, Auto ISO etc...). So I have to setup my camera each time manually for it to work with the settings I want (it works, using "P" mode).
Would be nice if it were working with the preset settings. 
Not a big problem though. Thank you for making it available. Very nice!

joelovescash, 2018-07-26:
I was having a similar issue with my a6500, particularly annoying when I was shooting in darker conditions and it would automatically boost the ISO to unwanted levels.

bojanv55, 2019-07-10:
I guess my pull request will solve also this one, since I changed this one:

`final Camera.Parameters params = cameraEx.createEmptyParameters();`

for this one 

`final Camera.Parameters params = cameraEx.getNormalCamera().getParameters();`

so instead creating new default empty params, use existing...

https://github.com/jonasjuffinger/TimeLapse/pull/23

jonasjuffinger, 2019-07-11:
Fixed with https://github.com/jonasjuffinger/TimeLapse/releases/tag/v1.1.0

closed 2019-07-11 by jonasjuffinger