jonasjuffinger/TimeLapse #22: Cannot Install app

darkGrimoire, 2019-03-31:
I'm using Sony A6300 with 2.01 Firmware version. Here's the error that I got (It is the same with all other OpenMemories app except the limit tweak)

```
Loading app list
Found 11 apps

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Model:              ILCE-6300
Product code:       0032828240
Serial number:      04532299
Firmware version:   2.01
Lens:               Model 0x18a08019 (Firmware 1.81)

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to app install mode
Waiting for camera to switch...
Downloading apk
Analyzing apk

Package: com.jonasjuffinger.timelapse
Version: 1.0.0

Starting task
Starting communication
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 76, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 291, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 84, in installApp
  File "C:\projects\sony-pmca-re\pmca\installer\__init__.py", line 84, in install
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
```

darkGrimoire, 2019-03-31:
And it still refuses even though I'm using MTP mode

jonasjuffinger, 2019-04-01:
Did you try it with another computer? It's probably better to ask questions about pmca here: https://github.com/ma1co/Sony-PMCA-RE. I cannot really help you, sorry.

closed 2020-01-03 by jonasjuffinger