jonasjuffinger/TimeLapse #26: Using Sony A6000 the app skips taking some photos

MartinArvidsson, 2019-07-27:
These are my current settings:
 Interval 1 sec
Shots 10
FPS 24
No silent shutter
No AEL
No BRC3

And the camera settings are F9 20sec. Single shooting.

What happends is that the camera takes the first picture, skips 2 and 3 takes 4 skips 5 takes 6 etc.

Does it have to do with the framerate selected?

MartinArvidsson, 2019-07-27:
It always ends up with half of the selected images, so if i say take 10 shots i end up with 5. If i say take 30 it ends up with 15

MartinArvidsson, 2019-07-27:
Sorry for spamming, I think i figured it out. If the interval is shorter than what the camera can manage to reload itself in, the app skips that image. because if i change the interval to 5 sec it takes all of the images.

MartinArvidsson, 2019-07-27:
My main goal is to take timelapses of stars so i can combine into long startrails, but since the app sometimes just stands there and skips it leads to it not taking a photo for 20sec since that is the shutter time (I think?) anyways, it leads to more of a dotted line in the sky than actual lines.

jonasjuffinger, 2019-07-27:
Hi Martin,
I just tried it with my 6300 and it takes all pictures. But it is possible that the 6000 is a little bit slower and therefore has problems.
For your specific purpose I would recommend a simple shutter cable that can be locked. Then the camera can take pictures all night long in continuous mode.

MartinArvidsson, 2019-07-28:
yea i will do that, thanks a lot!

closed 2019-07-28 by MartinArvidsson

MartinArvidsson, 2019-10-08:
Hi, I finally figured this out a few weeks ago, it seems that the built in "Long Exposure NR" was the cause of any issues since it tried taking a darkframe after each photo.

Turning this off allowed it to work flawlessly.