jonasjuffinger/TimeLapse #28: How can I download previous version of this lovely app.

ahmet1989, 2020-01-02:
Hi dear developers! I prefer older version. because I'm using this app with my phone's camera together. I set both to 5s intervalometer. The Phone is shooting wider photos and my nex-6 is shooting tele ones. they synced together well in 5 seconds timing mode. But in this new version there is no 5 seconds. It starts with 3,5 seconds after burst mode. Please help about this situation. Many thanks!!

ahmet1989, 2020-01-02:
Would you mind sending apk file to my e-mail address? You'll make me very happy if you can..

ahmetkonukseven1989@gmail.com

jonasjuffinger, 2020-01-03:
Hello Ahmet,
all older Versions can be downloaded here: https://github.com/jonasjuffinger/TimeLapse/releases
But does turning the dial also not allow to set 5s? This should work.

ahmet1989, 2020-01-03:
I didn't try turning the dial on las version but TimeLapse 1.2.0 works like a charm for me. :) many many thanks for this app and answering my question..

ahmet1989, 2020-01-03:
I didn't try turning the dial on last version but TimeLapse 1.2.0 works
like a charm for me. :) many many thanks for this app and answering my
question.. 🙏

Jonas Juffinger <notifications@github.com>, 3 Oca 2020 Cum, 11:53 tarihinde
şunu yazdı:

closed 2020-06-03 by jonasjuffinger