jonasjuffinger/TimeLapse #29: Inconsistent interval slider values

madcrow, 2020-01-19:
Hello,
I am running the app (v1.3.0) on my NEX-6 and observed the following:
When you move the interval slider and reach either of its extremities, the set of available values change.

At first launch, you are presented with the following values:
Burst-1.0s-4.0s-7.0s-10.0s-13.0s-16.0s-19s-24s-30s-36s-50s-80s-110sm-4.0m-7.0m-10m-16m-22m-28m-34m-[BLANK]
Once you have moved the slider to the rightmost position, the set of available values become:
Burst-3.0s-6.0s-9.0s-12.0s-15.0s-18.0s-22s-28s-34s-40s-70s-100s-3.0m-6.0m-9.0m-14m-20m-26m-32m-[BLANK] 
And when the leftmost end is reached, the value-set changes again, to:
Burst-3.5s-6.5s-9.5s-12.5s-15.5s-18.5s-23s-29s-35s-45s-75s-105s-3.5m-6.5m-9.5m-15m-21m-27m-33m-[BLANK]

Expected behavior:
-The value-set should be constant
-It should include the often used 1s, 2s, 3s and 4s options

A slightly unrelated question: what does the last (rightmost) label-less interval option mean?

jonasjuffinger, 2020-04-26:
Hi and sorry for the late answer.
Did you try setting the interval by turning the wheel? With this you should be able to select intervals in smaller steps.
The right most interval is actually a bug ^^

madcrow, 2020-04-26:
Hi there,
No problem at all.
Well, controlling the slider with the top dial is extremely weird :)
If I turn the wheel slowly, nothing happens, the slider value does not change. It does not matter if I turn it by 1 increment or 100. On the other hand, it seems to be sensitive of the speed of turning. If I turn the dial by a couple of increments really quick, then the slider indeed moves.
But due to this behavior, using the dial is not really suitable for fine adjustment in my case.

jonasjuffinger, 2020-04-26:
hmm that's really weird. Does the NEX-6 have a dial at the back next to the screen? This is the one I'm using on my a6300

madcrow, 2020-04-27:
Yes, it does. And in this app, it works identically to the top dial. If i turn it slowly, nothing happens. If I turn it really fast, it changes the value.