jonasjuffinger/TimeLapse #30: interval being ignored

elopez, 2020-02-24:
Hi! I'm trying to use this app in a Sony A6000, but no matter what I set in the interval slider, the app keeps taking photos continuously with no wait between them. I would expect that if I set the interval slider to, for instance, 10s, a picture would be taken every 10s and not immediately after the previous one. Do you have any ideas about what might be causing this? Thanks!

jonasjuffinger, 2020-04-26:
Hi and sorry for the late answer,
that's really weird. Did you try different shooting modes?

elopez, 2020-04-26:
Yeah, I tried many options. If you have any specific ideas on what to try, I can give it another go.

jonasjuffinger, 2020-04-26:
The app creates a log file on the SD Card in the TIMELPS folder. Could you delete it, if there is one already and then run the app for a few shots and upload the file? Then I can have a look.

elopez, 2020-04-26:
I just gave it another try and it seems to work now :thinking: maybe it was some configuration on the camera causing the trouble when I last tried? I'll close this issue for now but if it happens again I'll reopen it and get you that log. Thanks for the help! :smile:

closed 2020-04-26 by elopez