jonasjuffinger/TimeLapse #31: modify exposure in timelapse

PrJoss, 2020-03-28:
Hello!

Could opening / shutter / iso be modified once the timelapse has started? It would be a great option since sometimes it is necessary to correct the exposure (sunset, sunrise, others)

Thank you!

jonasjuffinger, 2020-04-26:
Hello PrJoss,
this is not possible right now, but I'll consider it for a next version.
For me it always worked quite well to use P mode when the light conditions are changing drastically.

PrJoss, 2020-04-26:
Thanks for taking it into account.  it's going to help a lot.  for timelapse it would help a lot in some particular cases

PrJoss, 2020-05-01:
> Hello PrJoss,
> this is not possible right now, but I'll consider it for a next version.
> For me it always worked quite well to use P mode when the light conditions are changing drastically.

Hello again.  just to add that in addition to being able to change the shutter, the diaphragm and the iso, it would also be great to see the camera's photometer.  Thank you!

pShota, 2021-05-01:
> Hello PrJoss,
> this is not possible right now, but I'll consider it for a next version.
> For me it always worked quite well to use P mode when the light conditions are changing drastically.

Hi jonas,
I was using this app in Aperture mode in sun rise scenario, and found out that the app did not take the exposure compensation setting (the right dial) into account, can this be fix??

j0n45h, 2022-08-30:
I forked this repo and added controlling the shutter speed and the aperture for my a6500 https://github.com/j0n45h/TimeLapse