jonasjuffinger/TimeLapse #32: Focus lock option

smensak, 2020-04-10:
Hi, 
I did recently discover that while using this app on my a6000, the focus cannot be locked. Even if I set the camera to manual mode before launching the app, it will still refocus between the shots. 

Is it possible to add focus lock switch to the app itself? It would be also useful for those timelapses from sunset into the night when you need to leave the exposure settings to the camera. 

If you want any more info/feedback, I'll be happy to help.

jonasjuffinger, 2020-04-26:
Hi smensak,
you are right, I never noticed that, because I often shoot time lapses with manual focus lenses.
I'll have a look into it and see if it is possible for the next version.
Thanks for your feedback!

jonasjuffinger, 2020-06-03:
I'm happy to announce that the new version 1.4 can now be set to manual focus.

smensak, 2020-06-03:
Thank you, great news! I'll test it soon.

smensak, 2020-06-05:
Happy to report that manual focus is indeed fully functional on my a6000.

closed 2020-06-05 by smensak