jonasjuffinger/TimeLapse #39: Print last picture taken as feedback

konfiot, 2020-10-07:
Hi,
First of all, awesome application, thanks !
When I run a timelapse, only a black screen is displayed with the progress of the timelapse, would it be possible to add the last taken picture under that information, in order to get a quick feedback on the process ?
Thanks

jonasjuffinger, 2020-10-09:
When setting the Auto Review setting, the picture is shown after every shot.

konfiot, 2020-10-11:
OK I might have missed that setting but I don't see it in the menu to configure the timelapse

konfiot, 2020-10-12:
Oh OK I see what you mean, but I have the auto review setting set to 5 seconds in the camera's menu, and all it displays is a black screen after taking each picture, even though it correctly saves a non-black picture

konfiot, 2020-10-12:
I'm using an A7s if that helps

DamarisUndDavid, 2021-01-16:
I'm having the same issue. I can't see the captured image. Tested all Auto Review settings. I had this app installed like a year ago and didn't have the issue. 
I'm using a sony a6000. Version 3.21

konfiot, 2021-04-24:
Hey, it's been a long time but I'm back on this
I downbgraded to 1.3.0 and it works !
I'm guessing that this line in the last commit may be the issue : https://github.com/jonasjuffinger/TimeLapse/commit/c785732b0d328b4dabd61580eeaa1006c5ce18ac#diff-da1b676f8ea8a7695bc89aca04cf2a9b0d1510823bfa3e92288cc5e50fce3d75R366
But I'm not an expert in android so I'm not sure, maybe I'll try to rebuild this without to test, not sure if I'll have the time

aquinox, 2025-06-12:
Are there any known bugs in 1.3.0? It gives me almost heart attack that after a while, checking there is just a black screen but when i want to check i have to reboot camera to verify if timelapse ran properly.